wimb0/home-assistant-saj-r5-modbus #34: Add `device_class` to sensor values

fsaris, 2021-04-19:
HA support device classes to give some more meaning to the sensor values: https://www.home-assistant.io/integrations/sensor/#device-class

Would be great if the `power`, `voltage`, `current` and `energy` sensors would report them self to HA with the `device_class`.

I'm willing to create a PR for this if you are open for this change.

wimb0, 2021-04-19:
Thanks good issue and pull request! This was something I wanted to implement, but did not do yet.

wimb0, 2021-04-19:
Merged PR #35 . Closing issue.

closed 2021-04-19 by wimb0

wimb0, 2021-04-19:
@fsaris Did you have specific reason to use `hass:sine-wave` instead of `mdi:sine-wave`?

fsaris, 2021-04-19:
only reason is that it is an icon from the default HA set.

wimb0, 2021-04-19:
> 
> 
> only reason is that it is an icon from the default HA set.

I wasn't aware there were built-in icons.
I did search for them, and could not find any official documentation about which ones there are.

fsaris, 2021-04-19:
ok, looks like we should go for the `mdi:` icons https://gist.github.com/thomasloven/01e03543364cf685eff969ad7980f5d7#gistcomment-3596355

wimb0, 2021-04-19:
I changed the `hass:` icons to `mdi:` in dev. will merge to master release a new version.

wimb0, 2021-04-19:
Fixed in release v1.3.1.