wimb0/home-assistant-saj-r5-modbus #46: Improve Energy dashboard compatability

wimb0, 2021-08-08:
I've been playing with the new Energy dashboard in Home Assistant 2021.8 and integration v1.5.0.
Apparently for the Solar production sensor to work, a new sensor which outputs kWh instead of W is needed.
It's not possible to select a live production value at the moment.
Will add a new sensor to fix this. Other ideas welcome!

[image]

fsaris, 2021-08-08:
The Total power generation works for me. It's in kWh

fsaris, 2021-08-08:
Is the sensor active? Looks like it's disabled in your screenshot.

[image]
Here it's correctly selectable

wimb0, 2021-08-08:
Yes, the sensor is active, but my Energy dashboard is weird.
[image]

The Total power generation is the total of what the Inverter has generated over its lifetime.
Maybe I'm wrong, but isn't the Energy dashboard based on live usage and production values?

My idea was to have high/low tariff values (From DSMR Reader via MQTT) as usage and Active power of inverter total output as production.

fsaris, 2021-08-08:
Here the info you should see in the developers tool

[image]

wimb0, 2021-08-08:
I guess I misunderstood how the Energy dashboard works, documentation is not really helpful.
I think you also have the meter totals as usage values?

wimb0, 2021-08-08:
[image]

Think I've got it working. Thanks @fsaris for the replies, they sent me in the right way.

I'll close this issue, as does not seem to be needed.

closed 2021-08-08 by wimb0