wimb0/home-assistant-saj-r5-modbus #59: Compatible with SAJ H1 Hybrid inverters

moreto84, 2022-03-01:
Hi,  Many thanks for your work.

Writting from spain, I got a H1 Hybrid inverter. Do you have any experience with this version?

I've been told that terminal supports Modbus, but I'd need to take out the Wifi Antenna and then I'd loose the statistics and the portal of the manufacturer. The issue is that it refreshes every 5 minutes and I'd need something with about 5-10 seconds.

Thanks!

wimb0, 2022-03-03:
Hi,

I do not know if this integration will work with your inverter.
At the moment it's only known to work with SAJ R5, Sununo and Suntrio inverters.
Can't find any modbus documentation for your H1 Hybrid inverter.
If you have documentation, maybe we can make the integration compatible.

Maybe it works, maybe it won't.
You can test what works.

manumora, 2022-05-13:
I think H1 doesn't work with RS485 TCP but maybe it can work with RS485 RTU, it is through RJ45 interface that provide H1 inverter. I am working on it.

wimb0, 2022-06-08:
Did you have any luck getting data out of the H1 inverter via modbus?

wimb0, 2022-08-19:
no reply. closing issue.

closed 2022-08-19 by wimb0