wimb0/home-assistant-saj-r5-modbus #64: Integration is using deprecated `DEVICE_CLASS_*` constants

frenck, 2023-01-03:
This custom integration uses deprecated `DEVICE_CLASS_*` constants in its codebase.

The `DEVICE_CLASS_*` constants have been deprecated and replaced in Home Assistant Core 2021.12 (over a year ago). I would highly suggest updating/migrating this integration to the new enums.

For example, for the device classes supported by the `sensor` platform, there is now a `SensorDeviceClass` enum. So if a sensor previously used the `DEVICE_CLASS_ENERGY` constant, it should now use `SensorDeviceClass.ENERGY`. Other platforms (like `binary_sensor`, and `number`) provide similar enumerations for their supported device classes.

The migration thus only consists of replacing constants with an enumeration member and is, therefore, very low impact and should be fairly straightforward.

If I can help resolve any questions regarding this change or migration, feel free to ask or respond to this issue. I'm happy to help!

Kindest regards,

../Frenck

wimb0, 2023-01-04:
Thanks. I did not know this.
Will fix asap.

closed 2023-01-06 by wimb0

frenck, 2023-01-06:
Awesome! Thanks, @wimb0 👍 

Happy New Year 🥂 

../Frenck

wimb0, 2023-01-06:
Fixed by @fsaris in #65 
Thanks!

commit 25da7c44 (2023-01-06): Update deprecated DEVICE_CLASS_* constants

Fixes: #64