wimb0/home-assistant-saj-r5-modbus #67: Values are frequently 'unavailable'

IngmarVerheij, 2023-04-19:
The integration works pretty well with the supplied instructions but the values are frequently set to 'unavailable' instead of the most recent read values. It seems its trying to read the modbus data to frequent and therefore fails, but this is a wild guess.

In the debug log I see the following ERROR frequently

`023-04-19 09:42:41.445 ERROR (MainThread) [custom_components.saj_modbus.hub] Unexpected error fetching SAJ data: unpack requires a buffer of 4 bytes
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 250, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/saj_modbus/hub.py", line 79, in _async_update_data
    realtime_data = await self.hass.async_add_executor_job(
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/saj_modbus/hub.py", line 150, in read_modbus_realtime_data
    faultMsg0 = decoder.decode_32bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 411, in decode_32bit_uint
    handle = self._unpack_words(fstring, handle)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 361, in _unpack_words
    handle = unpack(up, handle)
struct.error: unpack requires a buffer of 4 bytes
2023-04-19 09:43:41.684 INFO (MainThread) [custom_components.saj_modbus.hub] Fetching SAJ data recovered`

Is it possible to get a more detailed log file?

PS: Running on a SAJ inverted (Zonneplan provided) using an EW11A

wimb0, 2023-04-24:
I think I'm not able to help with this issue soon, as I moved and do not have Zonneplan solar panels and the SAJ inverter anymore.
However, I did not see these errors in my logfiles when I still had them.
Did you set a different scan interval, than the default 60?

IngmarVerheij, 2023-04-25:
Ah too bad. I’ll need to debug it myself then.
Where can I configure the scan interval?

Verzonden vanuit Outlook voor iOS<https://aka.ms/o0ukef>

newfilip, 2023-06-22:
where can i change the IP later?

wimb0, 2023-08-15:
You can't, HA does not support this.
I think you can edit **homeassisant\config.storage\core.config_entries** and change the IP there. Restart HA later.
Maybe you can also remove the integration and re-add it with the same Prefix.

closed 2023-08-15 by wimb0

Timmmeeeeh, 2025-03-19:
Hi All, 
I had the same issue and increased the polling which helped a bit. However I still ket getting the unavailable values for the active_power_of_inverter_total_output, making my graphs and calculations messy.
Only today I figures out that the l1_power reports the same value as the other sensor but RETAINS its last value in case of disconnection. For me using this sensor solved my porblems so i thought lets post it here, might be useful for sonmeone else.

wimb0, 2025-03-19:
Could the be the same issue as https://github.com/wimb0/home-assistant-saj-r5-modbus/issues/88 ?