wimb0/home-assistant-saj-r5-modbus #69: Config flow could not be loaded error message when trying to add integration

saberone, 2023-06-30:
I just installed the saj-modbus plugin. But when I try to add a SAJ Inverter Modbus integration I get the following error:

```Config flow could not be loaded: {"message":"Invalid handler specified"}```

When I check in the log file I see the following to error logs:

```
Logger: homeassistant.loader
Source: lsiopy/lib/python3.11/site-packages/pymodbus/client/sync.py:183
First occurred: June 27, 2023 at 12:23:38 (2 occurrences)
Last logged: 11:07:40

Unexpected exception importing platform custom_components.saj_modbus.config_flow
Traceback (most recent call last):
  File "/lsiopy/lib/python3.11/site-packages/homeassistant/loader.py", line 833, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lsiopy/lib/python3.11/site-packages/homeassistant/loader.py", line 850, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1126, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/saj_modbus/__init__.py", line 16, in <module>
    from .hub import SAJModbusHub
  File "/config/custom_components/saj_modbus/hub.py", line 10, in <module>
    from pymodbus.client.sync import ModbusTcpClient
  File "/config/lsiopy/lib/python3.11/site-packages/pymodbus/client/sync.py", line 179, in <module>
    class ModbusTcpClient(BaseModbusClient):
  File "/config/lsiopy/lib/python3.11/site-packages/pymodbus/client/sync.py", line 183, in ModbusTcpClient
    def __init__(self, host='127.0.0.1', port=Defaults.Port,
                                              ^^^^^^^^^^^^^
AttributeError: type object 'Defaults' has no attribute 'Port'
```

and

```
Logger: homeassistant.config_entries
Source: config_entries.py:863
First occurred: June 27, 2023 at 12:23:38 (2 occurrences)
Last logged: 11:07:40

Error occurred loading flow for integration saj_modbus: Exception importing custom_components.saj_modbus.config_flow
```

I'm running Home Assistant 2023.6.3 and HACS Integration version 1.32.1 and Frontend version: 20220906112053.

Any help is appreciated.

riconekkers, 2023-07-03:
same problem  here 

 Deze fout is ontstaan door een aangepaste integratie.

Logger: homeassistant.loader
Source: custom_components/saj_modbus/hub.py:10
Integration: SAJ Inverter Modbus (documentation, issues)
First occurred: 19:57:21 (3 occurrences)
Last logged: 19:58:57

Unexpected exception importing platform custom_components.saj_modbus.config_flow
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 833, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 850, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen imp
ortlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1126, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/saj_modbus/__init__.py", line 16, in <module>
    from .hub import SAJModbusHub
  File "/config/custom_components/saj_modbus/hub.py", line 10, in <module>
    from pymodbus.client.sync import ModbusTcpClient
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/sync.py", line 179, in <module>
    class ModbusTcpClient(BaseModbusClient):
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/sync.py", line 183, in ModbusTcpClient
    def __init__(self, host='127.0.0.1', port=Defaults.Port,
                                              ^^^^^^^^^^^^^
AttributeError: type object 'Defaults' has no attribute 'Port'

wimb0, 2023-07-04:
I cannot reproduce this issue with my own install.
Can you tell me which Home Assistant versions and installation type you are running, and on what kind of hardware?

riconekkers, 2023-07-04:
Home Assistant 2023.6.3
Supervisor 2023.06.4
Operating System 10.3
Frontend-versie: 20230608.0 - latest

I'm running a virtualbox image  on a server  with a AMD Ryzen 5 3600 6-Core Processor.

i'm testing now on a clean install  if i can reproduce the issue

riconekkers, 2023-07-04:
looks like solaredge modbus (https://github.com/binsentsu/home-assistant-solaredge-modbus) in combination with this  intergration gives this issue. If i remove the solaredge intergration and reboot HA the problem is gone. install solaredge intergration and  the problem is back.

saberone, 2023-07-04:
> looks like solaredge modbus (https://github.com/binsentsu/home-assistant-solaredge-modbus) in combination with this intergration gives this issue. If i remove the solaredge intergration and reboot HA the problem is gone. install solaredge intergration and the problem is back.

Hmm. I also have a solaredge modbus hacs module installed. So that combination triggers this issue.

wimb0, 2023-08-28:
Can you check if this problem is resolved by using version v1.8.0 ?
There were some changes to support newer pymodbus versions:
https://github.com/wimb0/home-assistant-saj-modbus/releases/tag/v1.8.0

saberone, 2023-08-28:
I installed version 1.8.0. And the installation works! Thanks for the update.

wimb0, 2023-08-28:
Thanks for checking. I will close this issue,

closed 2023-08-28 by wimb0