wimb0/home-assistant-saj-r5-modbus #72: Problems after adding the SmartMeter

Arise, 2023-09-20:
I've added a smart meter and a Saj SEC device, basically another Modbus reader.

The SEC device and the Elfin are both connected to the inverter RS485 lines. The Modbus addon works, now in Home Assistant I see the Inverter working, but it has many disconnects/communication blackouts.

Any idea how I should/could debug this?

[image]

SAJ Inverter status changed to Normal
7:06:17 PM - 3 minutes ago
SAJ Inverter status became unknown
7:05:47 PM - 3 minutes ago
SAJ Inverter status changed to Normal
7:03:02 PM - 6 minutes ago
SAJ Inverter status became unknown
7:02:32 PM - 6 minutes ago
SAJ Inverter status became unavailable
7:02:17 PM - 7 minutes ago
SAJ Inverter status changed to Normal
6:59:37 PM - 9 minutes ago
SAJ Inverter status became unknown
6:59:22 PM - 10 minutes ago
SAJ Inverter status became unavailable
6:59:07 PM - 10 minutes ago
SAJ Inverter status became unknown
6:58:52 PM - 10 minutes ago
SAJ Inverter status changed to Normal
6:55:27 PM - 13 minutes ago
SAJ Inverter status became unknown
6:54:42 PM - 14 minutes ago
SAJ Inverter status became unavailable
6:54:27 PM -

Arise, 2023-09-21:
In my case is an Elfin EE11a (the Ethernet version). Is connected through an ethernet cable. 4 wires are for ethernet, 4 wires for connecting to the inverter (from which 2 for RS485 and 2 for +7V and GND).

Arise, 2023-09-22:
Another update:
I'm pinging the Elfin device and it replies me continuously, so is not like the Elfin lost power or disconnects from the network.
Also I have the PHP REST integration with the inverter active (IP/status/status.php) and I receive the data every 30 seconds, so the inverter wifi module/the inverter itself works.
And the native SAJ Solar Inverter integration (platform: saj) also seems to work just fine, no hiccups.

wimb0, 2023-09-22:
Do you see anything in the logfiles?

Arise, 2023-09-22:
I don't know where to look actually.
If I look in standard Logbook available in HA interface I see this:
```
SAJ Inverter error message
2:23:43 PM - 5 minutes ago
SAJ Inverter working mode changed to 2
2:23:43 PM - 5 minutes ago
SAJ Inverter status changed to Normal
2:23:43 PM - 5 minutes ago
SAJ Inverter error message became unknown
2:23:28 PM - 5 minutes ago
SAJ Inverter working mode became unknown
2:23:28 PM - 5 minutes ago
SAJ Inverter status became unknown
2:23:28 PM - 5 minutes ago
SAJ Inverter error message became unavailable
2:23:03 PM - 6 minutes ago
SAJ Inverter working mode became unavailable
2:23:03 PM - 6 minutes ago
SAJ Inverter status became unavailable
2:23:03 PM - 6 minutes ago
SAJ Inverter error message became unknown
2:22:18 PM - 7 minutes ago
SAJ Inverter working mode became unknown
2:22:18 PM - 7 minutes ago
SAJ Inverter status became unknown
```

wimb0, 2023-12-04:
Can you test if your issue is solved by installing v1.9.0b1?
There were some modbus changes in recent HA versions, maybe this is related.
The integration is updated to a newer pymdbus version.

Arise, 2023-12-05:
Thanks, I will try after updating HA, should report back.

Arise, 2023-12-09:
Tested with 1.9.0 and I still have the problem with it going unknown.
How can I debug this? In which log file should I look to understand why it goes offline?

SAJ Inverter error message became unknown
11:16:34 AM - 5 seconds ago
SAJ Inverter working mode became unknown
11:16:34 AM - 5 seconds ago
SAJ Inverter status became unknown
11:16:34 AM - 5 seconds ago

wimb0, 2023-12-12:
It seems to me that the SEC device and the Elfin are both trying to be Modbus master. 
That does not work. It has nothing to do with this integration.
Maybe you can configure the SEC device to also be a modbus TCP server, so that you query the with this integration instead of the Elfin?

Arise, 2023-12-13:
I don't think that the SEC device can work as a TCP server. The SEC device is a pure modbus device when talking with the inverter and the smart meter. To both it uses separate pairs of wires.
I don't see any errors/complains on the SEC side, but I'm sure that whatever it exports to SAJ website is not 100% accurate, the real values are a bit less than it should.

The integration works but with hiccups. Every couple of minutes it loses connection then it reconnects. Because of that I believe is not measuring very accurately.

I will try to add some resistors at the ends (not sure which are my ends because of the way this is wired).

From the sec device I go with UTP CAT5e cable on the brown twisted pair dedicated for modbus transmission. After 30 meters of cable (routed outdoor) it reaches my Elfin EE11a and is connected through this connector.
[image]

Then in the above connector I go for another 1meter to the inverter with CAT5 cable (4 wires - voltage for elfin, ground and 2 wires for RS485).

So the elfin connector is a bit in the "middle" (more likely close to the inverter - 1meter cable) between the inverter and SEC device.

I was hoping for the inverter to "talk" and SEC and Elfin to listen.

When I had only the Elfin there was no issues with "listening", but now that I also have the SEC on the wires there is a offline problem every couple of minutes.

wimb0, 2023-12-13:
The Elfin does not only listen.
The integration asks for the data it wants via modbus.
I think the SEC device does the same.

Maybe, a modbus splitter or repeater can help.

wimb0, 2024-02-19:
Can't fix this.

closed 2024-02-19 by wimb0