wimb0/home-assistant-saj-r5-modbus #80: UI locks when communication time-outs occur

Lvdvvdk, 2024-03-21:
Hi,
I am using the modbus integration for my SAJ R5 inverter and utilize the Elfin EW-11A as interface.
During the day multiple communication time-outs occur which are logged as follows:

_2024-03-18 16:18:11.116 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 5.014 seconds (success: True)
2024-03-18 16:19:16.109 ERROR (SyncWorker_13) [pymodbus.logging] Connection to (192.168.50.233, 5555) failed: timed out
2024-03-18 16:19:16.112 ERROR (MainThread) [custom_components.saj_modbus.hub] Reading realtime data failed! Inverter is unreachable_

What could be the reason of the timeouts?
I have connected the EW11 as described in the README with the SAJ providing the power. Is it better to use an external power supply?
When time-outs are happening the HA UI is partly frozen.

Thanks

HindrikDeelstra, 2024-03-22:
It's probably not caused by this integration, but by WiFi connection issues.

When I started using this integration, SAJ R5 Modbus via an Elfin EW10, I initially had similar troubles. My inverter is located in the attic, and the interference of the solar DC cables seemed to be just enough to interrupt the WiFi communications. I had placed the cable alongside these cables for convenience.

To solve this, I re-arranged the orientation of the EW10 unit and specifically the connecting cable, so no part of it runs parallel or near to the solar DC cables. This has fixed the issue for me, I've not seen any WiFi disruptions since.

If re-arranging the cable trajectory does not help for you, you could try to improve your WiFi signal by adding or moving an AP closer to your EW11 unit.

Lvdvvdk, 2024-03-22:
> It's probably not caused by this integration, but by WiFi connection issues.
> 
> When I started using this integration, SAJ R5 Modbus via an Elfin EW10, I initially had similar troubles. My inverter is located in the attic, and the interference of the solar DC cables seemed to be just enough to interrupt the WiFi communications. I had placed the EW10 and its cable alongside the solar DC cables, for convenience.
> 
> To solve the connectivity issues, I re-arranged the orientation of the EW10 unit and the connecting cable, so no part of it runs parallel or near to the solar DC cables. This has fixed the issue for me, I've not seen any WiFi disruptions since.
> 
> If re-arranging the cable trajectory does not help for you, you could try to improve your WiFi signal by adding or moving an AP closer to your EW11 unit.

Thanks for your advice
The distance between the EW11 and AP is ~2m but I shall try to reposition the EW11 to see if that has any effect.
Can you clarify why the communication time-outs freeze the UI?

wimb0, 2024-04-02:
Are you sure the UI freezes are caused by this integration?
I've never seen this behaviour before.

wimb0, 2024-04-08:
I am not sure why this happens.
If the freezes are occuring when the EW11 is offline, everyone using this integration should have this issue at night,
when the inverter is offline. You are the first to mention this.
I have not seens this issue myself.
Maybe someone else can verify this?

wimb0, 2024-04-25:
I am closing this issue, as it is not caused by this integration.

closed 2024-04-25 by wimb0