wimb0/home-assistant-saj-r5-modbus #85: Setting Data Bits on Elfin EW10

ltv8, 2024-04-23:
Hello,

Maybe this is not really an issue with the integration itself, but I do not really know another place where I could possibly find help. 

I am trying to use the Serial TCP / RS232 connection method via USB with the Elfin EW10A. 
As stated in the integration, I would like to set the Data Bit value to 1. However, the Elfin EW10A only gives 2 options, 7 and 8. 
(I guess as a result of this), I am not getting any data in Home Assistant. 

Am I doing anything wrong? Or what could solve this? 

Many thanks in advance for any reply! 

Kind regards,
LTV8

wimb0, 2024-04-25:
It is not possible to set the Data Bit to 1, only 7 or 8.
Only the Stop Bit can be 1.

closed 2024-04-25 by wimb0

Erikovic, 2024-04-25:
I have two Zonneplan (SAJ) inverters, each with an EW10A. They work very well. The only issue is that one needed an external power supply. Are you sure the EW10A is active when it is connected to the inverter (do the lights come on)? My settings are as follows:
Baud Rate: 115200
Data Bit: 8
Stop Bit: 1

One tip is to avoid touching the new slider in HA. After doing so, I lost the peak in both inverters. I had to modify the code so that communication was set from 100% to 110%. In both cases, it was apparently set to 110%, while the slider did not go beyond 100. It will work fine until you adjust the slider yourself or restart HA.

wimb0, 2024-04-26:
I have updated the README in commit https://github.com/wimb0/home-assistant-saj-modbus/commit/2e7f0bbf450bdad50d51e4100fa0be3adc309ab6
For the EW10A the Data Bit must me set to 8, but the docs said 0 for some reason. This is now corrected.