wimb0/home-assistant-saj-r5-modbus #86: Setting the power slider to 100% is too low

Erikovic, 2024-04-25:
I have two SAJ inverters (via Zonneplan). I am very happy with the new Power Slider because I have issues with peak voltage (so I can now prevent it by using the Power Slider automatically). Both inverters easily reached the peak (Power Slider showed 100%), until I decided to actively use the Power Slider. When I set it to 100%, the peak was significantly lower for both inverters (luckily I don't have too many issues with peak voltage and can still reach most of the inverter's peak).

I have been working on the script myself to multiply the value passed through by 1.1. With that, I got the peak back for both inverters. The value behind the Power Slider was also set to 110%, even though the slider only goes up to 100%. Apparently, the inverters themselves are protected from going over their peak, and that protection is set higher than 100%. In short, I think many users lose their peak due to the Power Slider. If I experienced it with 2 out of 2 without ever doing anything myself, it might be a common problem. It could be a good solution for the overloaded grid, but I would prefer to leave that solution to the grid operator.

In addition, I have set the update time in the script to 10 seconds myself (it might even be possible to lower it further). This allows me to respond more quickly to peak voltage and greatly reduces the risk of outages. When the voltage goes above 251 volts, I decrease the power slider by 10% (1.1 times is 11%), and when it goes below 250 volts, I add 10% back. So far, this has prevented any outages, with three automatic 10% interventions to date. But the real test will come when the rain finally stops.

It would be great if you could more easily adjust the update time somewhere. But of course, that is just a luxury request.

mvn23, 2024-04-25:
It would be great to have some feedback from SAJ regarding this setting. They seem to be happy to let users set it to 110% through their app, however I'm still seeing peaks of 110% rated power despite explicitly setting it to 100%. It will be easier to test and verify once the sun will come out a bit more, but it definitely doesn't seem limited at 4000W with the current setting.

The higher output does seem limited by something, as I haven't seen the inverter report it for more than 1 interval (10 seconds) before dropping back down quite a lot. It's happy to sit around 4000W for longer periods of time.

Erikovic, 2024-04-25:
If I understand you correctly, you are also consistently around 4000W, so you are experiencing the same issue (i don't recognize your very short peak). If you happen to have the same type (R5-4K-S2), then it is likely set to 110% in your case as well. Before I had the Power Slider, the peak was always 4400W. So, Zonneplan (in my case, and probably for most users) delivers it at more than 100% by default. The protection is undoubtedly separate from the Power Slider at 4400W (because that number is very exact).

mvn23, 2024-04-25:
Looking at a bit more history I see the following events with the latest version of the integration and thus with the limit set at 100%:

[image]
Peak at exactly 4400W, with a drop immediately afterwards.

However, contrary to what I stated earlier, it does also still deliver power >100% for longer periods. Just not at exactly 4400W, but slightly below that value. This could of course be due to a lack of sun:
[image]

In short, I cannot reproduce the issue you're experiencing until now. There is one more possibility I'd like to explore, but it requires more sun than I have at home at the moment.

Arnoudd, 2024-04-25:
I also have the R5-4K-S2 and I do recognize this issue. I've got the feeling the inverter started in the morning with a correct limit of 4400 watt. 

But after playing around with the new limiter the limit seemed to remain much lower although I set it to 100%. The inverter then maxed to the 4000 watt area, don't know the exact numbers anymore. 

That's why I also changed the multiplier from 10 to 11. After that I could set the limit back to 4400 again.

mvn23, 2024-04-25:
I'm running some tests at the moment to verify the behavior. I'll work on a PR once I have a clearer idea of how it all behaves.

@wimb0 I realize we decided to keep the limit at 100% in #82, but if the inverter reverts to a default of 110% every morning on startup I don't see why we shouldn't allow that as well. What do you think?

wimb0, 2024-04-26:
Interesting discussion.
@mvn23 I agree, we should allow 110%, if it is the default SAJ sets on startup.
Is there a way to read the current value from the inverter?

(Sorry I haven't been this active in keeping the integration up to date, as I don't have an SAJ inverter anymore)

mvn23, 2024-05-05:
> Is there a way to read the current value from the inverter?

Unfortunately there does not seem to be a way to do this. I'm still testing the behavior (setting it to 100% explicitly every morning when operating mode switches to `Normal`) but there hasn't been enough sun to reach peak power reliably at my house.
When I have confirmed the suspected behavior I'm thinking to implement something like this:
* When switching operating mode to `Normal`, reset the power limit to `110%`.
* When switching operating mode away from `Normal`, give the power limit entity the state `unknown` or `unavailable`.

I'm still unsure on what to do with operating mode `Waiting`, so I'll need to test that as well.
Feedback on these plans is of course welcome.

closed 2024-06-17 by wimb0

HindrikDeelstra, 2024-06-18:
Just seeing this, but the SAJ R5-4K inverter is rated at exactly 4kW (4000W), *not* 4400W! The limit of 100% corresponds to the correct maximum power of 4000W.

Setting it to 110% is overclocking"" the unit. I do not know if this will void warranty, or could cause issues, but it does not seem logical or a sane default to me.

There's a difference between real power (in Watts), and apparent power (which is measured in VA). For DC loads, these values and physical meaning is the same, but not so for AC loads!

AC loads can present a different power than is actually usefully used. Due to a skew in the angles between the electrical and magnetic waves,  some apparent power is developed, which is not actual, useful power. This is represented by the "power factor", or cos(phi).

The SAJ R5-4K inverter is rated for 4000W actual power, or 4400VA, apparent power. It is *not* rated for 4400W actual power, as can be seen on the product label. The "power limit"setting sets the desired *actual* power, so 4000W max.

mvn23, 2024-06-18:
Please read the discussion above. Tests have shown that the inverter reverts to a default of 110% on every restart **even if we do not explicitly set a value**. This is the reason we have now changed the maximum (and default on initialization) to that value. I was skeptical about this as well when I saw the specs of the inverter, but after testing actual behavior I have changed my mind and implemented the 110%.

PR #89 (2024-06-17): Update Limit Power entity to match actual device behavior
After running some tests it seems that the inverter resets its output limit to `110%` every time `mpvmode` changes to `2` (`Normal`). This PR is meant to reflect that behavior in the following ways:
* When `mpvmode` is not `2`, disable the `Limit Power` number entity
* When `mpvmode` changes to `2`, set the `Limit Power` number entity to `110%` (internally only, no data is sent to the inverter)
* While `mpvmode` remains `2`, track the power limit internally

One potential issue with this method is that the `Limit Power` entity may de-sy [... cut]