wimb0/home-assistant-saj-r5-modbus #87: Extra details from a frequent error

Arise, 2024-04-27:
I get this issue in the logs, but it could be related with the other issue I've mentioned in the past.

```
This error originated from a custom integration.

Logger: custom_components.saj_modbus.hub
Source: helpers/update_coordinator.py:315
integration: SAJ Inverter Modbus (documentation, issues)
First occurred: 7:30:35 PM (1 occurrences)
Last logged: 7:30:35 PM

Unexpected error fetching SAJ data: unpack requires a buffer of 4 bytes
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/saj_modbus/hub.py", line 95, in _async_update_data
    realtime_data = await self.hass.async_add_executor_job(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/saj_modbus/hub.py", line 166, in read_modbus_realtime_data
    faultMsg0 = decoder.decode_32bit_uint()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/payload.py", line 380, in decode_32bit_uint
    handle = self._unpack_words(fstring, handle)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/payload.py", line 340, in _unpack_words
    handle = unpack(f"!{wc_value}H", handle)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes
```

Arise, 2024-04-27:
I've enabled logs and watched with `tail -f /config/home-assistant.log` and when it had problems, this is how it went:

```
...
2024-04-27 19:47:25.318 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.275 seconds (success: True)
2024-04-27 19:47:35.318 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.277 seconds (success: True)
2024-04-27 19:47:45.321 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.279 seconds (success: True)
2024-04-27 19:47:55.322 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.280 seconds (success: True)
2024-04-27 19:48:05.324 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.283 seconds (success: True)
2024-04-27 19:48:15.324 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.283 seconds (success: True)
2024-04-27 19:48:25.134 ERROR (MainThread) [custom_components.saj_modbus.hub] Unexpected error fetching SAJ data: unpack requires a buffer of 4 bytes
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/saj_modbus/hub.py", line 95, in _async_update_data
    realtime_data = await self.hass.async_add_executor_job(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/saj_modbus/hub.py", line 166, in read_modbus_realtime_data
    faultMsg0 = decoder.decode_32bit_uint()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/payload.py", line 380, in decode_32bit_uint
    handle = self._unpack_words(fstring, handle)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/payload.py", line 340, in _unpack_words
    handle = unpack(f"!{wc_value}H", handle)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes
2024-04-27 19:48:25.139 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.098 seconds (success: False)
2024-04-27 19:48:40.047 INFO (MainThread) [custom_components.saj_modbus.hub] Fetching SAJ data recovered
2024-04-27 19:48:40.047 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 5.007 seconds (success: True)
2024-04-27 19:48:55.051 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 5.009 seconds (success: True)
2024-04-27 19:49:05.310 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.269 seconds (success: True)
2024-04-27 19:49:15.310 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.269 seconds (success: True)
2024-04-27 19:49:25.311 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.269 seconds (success: True)
...
```

Arise, 2024-04-27:
My feeling is this could be related to the SEC device, if I understand properly this modbus thingie, the inverter is the slave and the Elfin and the SEC smart meter devices are the masters, and both harass the inverter requesting data.

Maybe at some point the SEC is requesting data, inverter replies and the Elfin device is catching the response on the fly, tries to decode using his known structure, the structure is not matching because the reply was meant for the SEC device, so the SAJ integration throws the exception and then it takes some time to recover (like 20 seconds?!?)

After the "Fetching SAJ data recovered" there are also 2 long time processing jobs reports, which is strange.

mvn23, 2024-04-27:
See https://github.com/wimb0/home-assistant-saj-modbus/issues/72#issuecomment-1851652904 for the most likely cause of your troubles. You **can not connect more than one device** that requests data from your inverter's RS485 port and expect it to work without issues. Modbus is not designed for that.

Arise, 2024-04-27:
I understand the part with "not expect it to work without issues". But most of the time the integration works fine, only from time to time the problem appears in which case handling the exception thrown could maybe improve the behavior?

Arise, 2024-05-13:
I ordered two "gateways" units, something that works like 2 masters and 1 slave and I hope to see if this is going to fix my issues.
I will report back with the outcome, and I will propose closing the issue.

wimb0, 2024-05-27:
Good to hear the issue is solved.
Which gateway did you use?

Arise, 2024-05-27:
The ones that appear near the elfins in the image.

Two Master One /Multiple Slave RS485 Repeater Cache Optoelectronic Isolation Modbus Dual Master Hub
https://a.aliexpress.com/_EwiBqvz

closed 2024-05-27 by wimb0