wimb0/home-assistant-saj-r5-modbus #88: Issue while calculating the PV production

Arise, 2024-04-27:
I've noticed that from time to time, even without an error described in other bug reports my [tesla-style-solar-power-card](https://github.com/reptilex/tesla-style-solar-power-card) shows **0** for the PV production. 

Not sure if there is something in how I compute the sensors, but this is the problem I see frequently when the saj modbus integration takes longer time (like multiple 5 seconds) to process the message.

[image]

mvn23, 2024-04-27:
Disconnect your SEC and see if the problems persist...

Arise, 2024-05-27:
Issue fixed as well.

closed 2024-05-27 by Arise