wimb0/home-assistant-saj-r5-modbus #90: Add option for modbus to usb converter

manusjevanalles, 2024-06-24:
Add option for modbus to usb converter.

Currently, there is only the option to use a modbus to ethernet converter when using this integration. I am reading the modbus values using a modbus to usb converter, directly plugged into the server running Home Assistant. Would it be possible to add the option to use a usb converter?

wimb0, 2024-07-02:
I am not sure if I can add this. Does the converter present itself as a serial port?
It may be a dirty workaround, but you could use something like ser2net.

manusjevanalles, 2024-07-03:
The converter is indeed presented as serial port, and added as following to modbus.yaml:
  - name: SAJ-modbus
    type: serial
    port: /dev/ttyUSB1
    baudrate: 9600
    bytesize: 8
    method: rtu
    parity: N
    stopbits: 1
    delay: 0
    message_wait_milliseconds: 30
    timeout: 5
    sensors:

wimb0, 2024-11-08:
Checked this issue again, I can't add this.
I suggest you use the normal modbus integration in Home Assistant and config the correct registers.

wimb0, 2026-09-18:
Hi, it took a while, but since version v4.1.0 this is now supported!

closed 2026-09-18 by wimb0