wimb0/home-assistant-saj-r5-modbus #91: Saj sununo plus help

gerben-f1, 2024-06-26:
Hello,
Can u help me to gett it work?

i have installed youre addon
I have an elfin EE10A
There are 2 connections on the elfin
Ethernet and a green connector with txd rxd + -
I connected an 9 pin rs232 connector to the saj inverter and the other end has 3 wires

connect the wires to the ground txd rxd
And takes a 5v power and connected the plus wire to + on green connector from elvin

but i cant read out my saj inverter what do i wrong?
[image]
[image]
[image]

[image]

wimb0, 2024-07-02:
Have you tried reversing the TXD and RXD? TX from Elfin needs to go to RX from SAJ. RX needs to go to TX.
Maybe this helps?

gerben-f1, 2024-07-03:
I changed rx and tx but not workingWhat are the correct settings on the elfinFor rs232 saj sununo plu 3k-mM.v.g Gerben