wimb0/home-assistant-saj-r5-modbus #92: Not closing the realtime connection causes intermitting data results

gerritjanf, 2024-07-16:
Love your module! Reading out my Zonneplan inverter with the EW11a has been a success.

For some time i have been noticing intermittend data gaps. Investigation shows that home assistant tries to re-use the previous connection and seems successfull but does not get any data returned. Next run a new connection will be invoked and the data is gathered without issue.

**Broken logging**
The ultra short times it took to gather resprent the data  gaps

```
2024-07-16 14:02:51.748 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.411 seconds (success: True)
2024-07-16 14:03:01.339 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.001 seconds (success: True)
2024-07-16 14:03:13.377 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 2.040 seconds (success: True)
2024-07-16 14:03:23.765 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.428 seconds (success: True)
2024-07-16 14:03:33.338 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.001 seconds (success: True)
2024-07-16 14:03:45.362 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 2.024 seconds (success: True)
2024-07-16 14:03:55.766 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.428 seconds (success: True)
2024-07-16 14:04:05.339 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.001 seconds (success: True)
2024-07-16 14:04:17.844 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 2.507 seconds (success: True)
2024-07-16 14:04:27.802 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.464 seconds (success: True)
2024-07-16 14:04:37.339 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.001 seconds (success: True)
2024-07-16 14:04:49.366 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 2.028 seconds (success: True)
2024-07-16 14:04:59.788 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.450 seconds (success: True)
2024-07-16 14:05:09.339 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.002 seconds (success: True)
2024-07-16 14:05:21.859 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 2.521 seconds (success: True)
2024-07-16 14:05:31.567 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.229 seconds (success: True)
2024-07-16 14:05:41.338 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.001 seconds (success: True)
2024-07-16 14:05:53.372 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 2.035 seconds (success: True)
2024-07-16 14:06:03.563 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.225 seconds (success: True)
2024-07-16 14:06:13.339 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.001 seconds (success: True)
```

**successfull logging**
```
2024-07-16 13:57:39.327 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.233 seconds (success: True)
2024-07-16 13:57:49.321 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.227 seconds (success: True)
2024-07-16 13:57:59.319 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.226 seconds (success: True)
2024-07-16 13:58:09.319 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.226 seconds (success: True)
2024-07-16 13:58:21.708 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 2.615 seconds (success: True)
2024-07-16 13:58:33.142 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 2.048 seconds (success: True)
2024-07-16 13:58:43.324 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.230 seconds (success: True)
2024-07-16 13:58:53.319 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.226 seconds (success: True)
2024-07-16 13:59:03.317 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.225 seconds (success: True)
2024-07-16 13:59:13.317 DEBUG (MainThread) [custom_components.saj_modbus.hub] Finished fetching SAJ data in 0.224 seconds (success: True)
```

The difference shows in Home Assistant history

I created a simple workaround myself in code as i could tweak the setting of the EW11a to fix the issue. However maybe this is not the best fix.

At line 91 of hub.py i included a call to close the connection so it will be new everytime. 
https://github.com/wimb0/home-assistant-saj-r5-modbus/blob/main/custom_components/saj_modbus/hub.py#L91
called method: `self.close()`

gerritjanf, 2024-07-16:
[image]

wimb0, 2024-07-27:
Thanks for reporting this!
I will look into it and add the fix in the next version.

wimb0, 2024-08-26:
I am trying to add this.
Can you post the exact changes you added?

gerritjanf, 2024-10-01:
I created a pull request to be exact on what i changed. It is now linked to this issue.

closed 2024-10-28 by wimb0