wimb0/home-assistant-saj-r5-modbus #95: Deprecated functions logged on startup

HindrikDeelstra, 2024-08-18:
I'm seeing some warnings at startup, using HomeAssistant Core 2024.8.2. I have no information on when this started, I just saw it by coincedence today.

At startup, the following messages are logged:

`WARNING (ImportExecutor_0) [homeassistant.helpers.typing] HomeAssistantType was used from saj_modbus, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'saj_modbus' custom integration`

and:

`WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'saj_modbus' calls async_forward_entry_setup for integration, saj_modbus with title: SAJ and entry_id: 16e7df1e06dde3536e9b716cfee04d38, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/saj_modbus/__init__.py, line 61: hass.async_create_task(, please create a bug report at https://github.com/wimb0/home-assistant-saj-r5-modbus/issues`

and:

`WARNING (MainThread) [homeassistant.helpers.frame] Detected code that calls async_forward_entry_setup for integration saj_modbus with title: SAJ and entry_id: 16e7df1e06dde3536e9b716cfee04d38, during setup without awaiting async_forward_entry_setup, which can cause the setup lock to be released before the setup is done. This will stop working in Home Assistant 2025.1. Please report this issue.`

Although the last line does not explicitly mention that the deprecated code is present in the saj_modbus component, but I've included it here just in case it is ;)

wimb0, 2024-08-26:
This issue should be fixed in pre-release version v2.2.2.
Can you test this?

HindrikDeelstra, 2024-08-26:
I've downloaded the 2.2.3 pre-release version, and restarted HA. I can report that the warnings posted earlier no longer appear in the logs ;)

All functionality seems to be working as far as I can tell at the moment, no immediate or obvious problems detected.

wimb0, 2024-08-27:
Thanks for checking. I will create a normal release with these fixes.

commit 0bccdb26 (2024-08-26): Update hub.py

Fixes #95 : WARNING (ImportExecutor_0) [homeassistant.helpers.typing] HomeAssistantType was used from saj_modbus, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'saj_modbus' custom integration
PR #96 (2024-08-26): Fixes deprecation warnings.
Fixes #95
closed 2024-08-27 by wimb0