wimb0/home-assistant-saj-r5-modbus #97: Deprecated constant POWER_VOLT_AMPERE_REACTIVE logged 2024.9

HindrikDeelstra, 2024-09-04:
Just updated HA to 2024.9, and this warning shows up in the logs:

`2024-09-04 21:14:02.329 WARNING (ImportExecutor_0) [homeassistant.const] POWER_VOLT_AMPERE_REACTIVE was used from saj_modbus, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'saj_modbus' custom integration`

wimb0, 2024-09-11:
I've changed this in the UnitOfReactivePower branch. Can you test this?
(Still don't have an SAJ Inverter :( )

HindrikDeelstra, 2024-09-11:
I've not found a way (yet?) to pull/checkout a branch from HA / HACS, but, since it was only 2 changes in 1 file, I've edited the file manually in my installation, and restarted HA.

The warning is now absent, and the sensor still works, displaying the correct unit of measurement ("var").

So, it looks like the change in the UnitOfReactivePower branch could be merged into main!

wimb0, 2024-09-11:
Thanks for reporting the issue and testing! I'll create a new release.

It seems HACS 2.0 removed the ability to choose a branch. Didn't know that.
See https://github.com/hacs/integration/issues/4009 and https://hacs.xyz/docs/use/entities/update/

wimb0, 2024-09-11:
Fixed in release v2.2.4 , PR #98

HindrikDeelstra, 2024-09-11:
> Thanks for reporting the issue and testing! I'll create a new release.
> 
> It seems HACS 2.0 removed the ability to choose a branch. Didn't know that. See [hacs/integration#4009](https://github.com/hacs/integration/issues/4009) and https://hacs.xyz/docs/use/entities/update/

I had missed that, but the good news is: it works! ;) I've just been able to download the version based on the full commit SHA. Couldn't test it based on the  branch name anymore, since it had already been merged ;)

Thanks!

wimb0, 2024-09-11:
v2.2.4 should now be available, had some issue creating the release.

HindrikDeelstra, 2024-09-11:
The 2.2.4 release was available, and was updated via HACS already, at around 14:14 o'clock this afternoon. So it looks like it did work the first time ;)