wimb0/home-assistant-saj-r5-modbus #101: Home Assistant 2025.1.0bx Setup failed for custom integration 'saj_modbus': Unable to import component: No module named 'pymodbus.register_read_message'

Fritz4all, 2024-12-25:
After upgrading to the latest Beta version 2025.1.0b1 & 2025.1.0b2

Setup failed for custom integration 'saj_modbus': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/saj_modbus/__init__.py", line 16, in <module>
    from .hub import SAJModbusHub
  File "/config/custom_components/saj_modbus/hub.py", line 2, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

AndreTzerra, 2025-01-04:
After updating to 2025.1 I got it to.

`Setup failed for custom integration 'saj_modbus': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/saj_modbus/__init__.py", line 16, in <module>
    from .hub import SAJModbusHub
  File "/config/custom_components/saj_modbus/hub.py", line 2, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'`

wimb0, 2025-01-04:
Thank you both for reporting this, I know what the issue is and will fix it as soon as possible.
Should have a new version available tomorrow.

wimb0, 2025-01-04:
Tried to fix the issue.
Pymodbus in HomeAssistant was updated to 3.7.4, and has a breaking change.
I've pinned the version to 3.6.9, which -should- fix the issue, according to other addons.

You can test with version 2.2.7.b1: https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.2.7b1

Arise, 2025-01-05:
Hi, it looks like pinning the version in the custom saj modbus integration breaks the standard modbus from HA, so it is one or the other, you can't have both.

lucwillems, 2025-01-05:
was somehow unable to switch to the v2.2.7b1 tag using hacs, so update the local manifestt.json file on disk to update the library version. after restart HA, all seems back to normal , entities are ready from inverter using modbus

Arnoudd, 2025-01-05:
> Hi, it looks like pinning the version in the custom saj modbus integration breaks the standard modbus from HA, so it is one or the other, you can't have both.

This is what I'm experiencing too. 
This SAJ modbus fix broke my Solis modbus integration. 

This is what I found when googling this: https://github.com/home-assistant/core/issues/134593

wimb0, 2025-01-05:
Thanks for the feeback.

I've just released v2.2.7b2 : https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/v2.2.7b2

This should make it compatible with pymodbus 3.7.4 which is in HomeAssistant 2025.1, instead of pinning an old version.

Arise, 2025-01-05:
Thanks for mentioning this.
I think this should be the end of the "python" based version and we should all move/transition to a stock/home assistant modbus integration, so we no longer stay dependent on python/this custom hacs integration.

Tested and I still get an error, this time on:
from pymodbus.pdu.register_write_message import ModbusResponse

Arnoudd, 2025-01-05:
> Thanks for mentioning this.
>
> I think this should be the end of the "python" based version and we should all move/transition to a stock/home assistant modbus integration, so we no longer stay [...]

Same here.

File "/config/custom_components/saj_modbus/hub.py", line 3, in <module>
    from pymodbus.pdu.register_write_message import ModbusResponse
ImportError: cannot import name 'ModbusResponse' from 'pymodbus.pdu.register_write_message' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/register_write_message.py)

Arise, 2025-01-05:
Because I don't write to modbus, I simple commented the import in hub.py
#from pymodbus.pdu.register_write_message import ModbusResponse

and replaced the function:
```
    def _write_registers(
        self, unit: int, address: int, values: list[int] | int
    ):# -> ModbusResponse:
        """Write values to registers."""
        with self._lock:
            return None
        #    return self._client.write_registers(address=address, values=values, slave=unit)
        
```

wimb0, 2025-01-05:
Fixed the  'pymodbus.pdu.register_write_message' error in v2.2.7b3

Arnoudd, 2025-01-05:
> Fixed the  'pymodbus.pdu.register_write_message' error in v2.2.7b3
> 
> 
> 
> 

Unfortunately I get a different error now: 

[code]

File "/config/custom_components/saj_modbus/hub.py", line 3, in <module>
    from pymodbus.register_write_message import ModbusResponse
ModuleNotFoundError: No module named 'pymodbus.register_write_message'
[/code]

wimb0, 2025-01-05:
> Thanks for mentioning this. I think this should be the end of the "python" based version and we should all move/transition to a stock/home assistant modbus integration, so we no longer stay [...]

These fixes are a workaround and permanent fix need to me made.
I've looked in to this, and I am not sure how to use the HA stock modbus integration for this custom integration.

This is an unfortunate breaking change in pymodbus, which was not listed in the HA releasenotes, and numerous other integration suffer the same problem.

Arise, 2025-01-05:
@wimb0 at some point I think I've started to play with the inverter, but in the end, I've finalized only the modbus integration for my DTSU (smart meter).

But it looks I also have some code for the inverter (modbus.yaml) (must be included in configuration.yaml as `modbus: !include modbus.yaml`), a sample of the modbus.yaml  is below:
```
- name: modbus_saj
  type: tcp
  host: 192.168.100.34
  port: 8899
  message_wait_milliseconds: 1000
  timeout: 5
  delay: 3
  sensors:
    - name: modbus_register_block
      slave: 1
      address: 256
      count: 36
      data_type: custom
      structure: ">36h"
      scan_interval: 5

    - name: modbus_inverter_name
      slave: 1
      address: 36611
      count: 10
      data_type: string
      scan_interval: 5

    - name: modbus_info_block
      slave: 1
      address: 36608
      count: 29
      data_type: custom
      structure: ">29h"
      scan_interval: 5
```
I just tested the sensor.modbus_inverter_name and it gives me back my R5 inverter name.

wimb0, 2025-01-05:
> @wimb0 at some point I think I've started to play with the inverter, but in the end, I've finalized only the modbus integration for my DTSU (smart meter).
>
> But it looks I also have some code for the [...]

That indeed works, but that gives a very large yaml file I think.

wimb0, 2025-01-05:
Apparently, the way this integrations is writing modbus registers is removed from pymodbus: https://github.com/pymodbus-dev/pymodbus/pull/2383

wimb0, 2025-01-05:
Release version 2.2.7b4: https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.2.7b4

This changes the way the integration reads en writes the modbus registers to a more simpeler/standard method.

Please test and report if it is working or not.

(I might be getting an SAJ R5 inverter again with a small solar array to test :-) )

Arnoudd, 2025-01-05:
Looks good. No errors here anymore and all values are loading.

Thanks for your work!

wimb0, 2025-01-05:
Nice! Good to hear. Thanks for testing.
Can you maybe also test if setting the inverter output power works or does not give errors?

Arnoudd, 2025-01-05:
> Nice! Good to hear. Thanks for testing.
> 
> Can you maybe also test if setting the inverter output power works or does not give errors?

I am able to change the number.saj_limit_power value without errors. 
It looks like it's working, but I can't test if it really enforces a power limit because of the lack of sun. 😆

wimb0, 2025-01-05:
Issue is fixed in release v2.3.0 : https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.3.0

closed 2025-01-05 by wimb0

Arnoudd, 2025-02-07:
> Issue is fixed in release v2.3.0 : https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.3.0

Oké, weird thing now is that today I noticed that I can't change the number.saj_limit_power value anymore. A bit weird since it was working when I last posted?

The error that I'm getting is:
Actie number/set_value kon niet worden uitgevoerd. object of type 'int' has no len()

Any idea what is going wrong here?