wimb0/home-assistant-saj-r5-modbus #103: Can't set the number.saj_limit_power anymore

Arnoudd, 2025-02-07:
Today I noticed that the limit power isn't adjustable anymore. If I trie to change the values I get the error:

**_Action number/set_value can not be excecuted. object of type 'int' has no len()_**

Any idea what is going wrong here?

I reverted to the previous ...b4 version but that didn't help. 

Could you please take a look?

Arnoudd, 2025-02-07:
Is this helpfull?
[code]
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 122, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 420, in async_set_native_value
    await self.hass.async_add_executor_job(self.set_native_value, value)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/saj_modbus/number.py", line 75, in set_native_value
    self.coordinator.set_value(self.entity_description.key, value)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/saj_modbus/hub.py", line 344, in set_value
    self.set_limitpower(value)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^
  File "/config/custom_components/saj_modbus/hub.py", line 319, in set_limitpower
    response = self._write_registers(unit=1, address=0x801F, values=int(value*10))
  File "/config/custom_components/saj_modbus/hub.py", line 77, in _write_registers
    return self._client.write_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, values=values, slave=unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 514, in write_registers
    return self.execute(no_response_expected, pdu_reg.WriteMultipleRegistersRequest(address=address, registers=values,dev_id=slave))
                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/pdu.py", line 36, in __init__
    self.count: int = count or len(self.registers)
                               ~~~^^^^^^^^^^^^^^^^
TypeError: object of type 'int' has no len()
2025-02-07 14:29:00.384 ERROR (MainThread) [homeassistant.components.automation.export_limit_saj] While executing automation automation.export_limit_saj
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/automation/__init__.py", line 717, in async_trigger
    return await self.action_script.async_run(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        variables, trigger_context, started_action
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )[/code]

I'm using the latest core version: 2025.2.0

gitproos, 2025-02-10:
I have this as well. In the interface the message is condensed to:

Actie number/set_value kon niet worden uitgevoerd. object of type 'int' has no len()

wimb0, 2025-02-10:
I took a quick look, but cannot really find out where or why exactly the error happens.
Not sure what recently changed Home Assistant regarding this.
Yes, and int has no len(), so maybe it requires a str or list now.

@mvn23 Do you have any idea what might be causing this? I think you wrote the limit power part.

mvn23, 2025-02-10:
I did indeed. Looks like `pymodbus` was bumped from `3.7.4` to `3.8.3` in Home Assistant `2025.2.0`, which is likely related. I will look into it.

wimb0, 2025-02-11:
Fixed by @mvn23 . Will be released in the next version later today.

wimb0, 2025-02-11:
Should be fixed in Release v2.3.2 
https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.3.2

Please report if it works (or not :D )

gitproos, 2025-02-11:
[image]

Works perfectly!

wimb0, 2025-02-11:
Thanks for response, I'll close the issue.

PR #107 (2025-02-11): Fix set_limitpower
Provide a list[int] as "values" argument to _write_registers

With the update to `pymodbus>=3.8.0` the accepted argument types for the `write_registers` method were reduced. We now need to provide a `list[int]`. This PR takes care of that.

fixes #103
closed 2025-02-11 by wimb0