wimb0/home-assistant-saj-r5-modbus #104: Deprecated function BinaryPayloadDecoder

gitproos, 2025-02-10:
Hi, in Home Assistant I see the following:

Logger: pymodbus.logging
Bron: custom_components/saj_modbus/hub.py:151
integratie: SAJ R5 Inverter Modbus ([documentatie](https://github.com/wimb0/home-assistant-saj-r5-modbus), [problemen](https://github.com/wimb0/home-assistant-saj-r5-modbus/issues))
Eerst voorgekomen: 07:58:16 (285 gebeurtenissen)
Laatst gelogd: 09:33:01
BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

Versions:
SAJ modbus integration 2.3.0
Core 2025.2.1
Supervisor 2025.02.0
Operating System 14.2
Frontend 20250205.0

wimb0, 2025-02-10:
I'll look in to it, thanks for reporting.
It is not an urgent issue, just a heads up that something will be deprecated in Pymodbus 3.9.0.

wimb0, 2025-02-10:
Made a fix / workaround in pre-release 2.3.1: https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.3.1
Other projects also use the same fix, so I think it should work here as well.

gitproos, 2025-02-10:
No rush, but that was fast! Would you like me to test the pre-release? I'm not sure how to install the pre-release though. I've installed the mod via HACS and that does not provide versions or betas to select it seems.

alexdelprete, 2025-02-10:
> Made a fix / workaround in pre-release 2.3.1: https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.3.1 Other projects also use the same fix, so I think it should work here as well.

I was looking for a quick solution for my ABB/FIMER integration, thanks for this. What other projects are using this workaround? Has it been tested and confirmed working?

Also, I noticed that in [pymodbus v3.8.3](https://github.com/pymodbus-dev/pymodbus/releases/tag/v3.8.3) the dev removed the deprecation in payload.py (see [PR #2532](https://github.com/pymodbus-dev/pymodbus/pull/2532)). Since HA 2025.x uses pymodbus v3.8.3, why do deprecation warnings show up in the log?

wimb0, 2025-02-11:
> No rush, but that was fast! Would you like me to test the pre-release? I'm not sure how to install the pre-release though. I've installed the mod via HACS and that does not provide versions or betas to select it seems.

If you go to HACS and choose reinstall the integration, you can expand different versions to install.

> > Made a fix / workaround in pre-release 2.3.1: https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.3.1 Other projects also use the same fix, so I think it should work here as well.
> 
> I was looking for a quick solution for my ABB/FIMER integration, thanks for this. What other projects are using this workaround? Has it been tested and confirmed working?
> 
> Also, I noticed that in [pymodbus v3.8.3](https://github.com/pymodbus-dev/pymodbus/releases/tag/v3.8.3) the dev removed deprecate from `payload.py` (see [PR #2532](https://github.com/pymodbus-dev/pymodbus/pull/2532)). Since HA 2025.x uses pymodbus v3.8.3, why do deprecation warnings show up in the log?

The following integrations use the same fix:
home-assistant-solaredge-modbus since v.2.0.1 : https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V2.0.1****
homeassistant-solax-modbus since v2025.01.9 : https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.9

I have no idea why they removed the deprecation warning, and why it still shows up.
Maybe its not handled in payload.py anymore but somewhere else?

gitproos, 2025-02-11:
Installed the pre-release version 2.3.1 via HACS and restarted the host. Inverter data is visible and I'm not seeing any deprecation messages in the logs. This looks resolved to me! Thank you @wimb0 !

wimb0, 2025-02-11:
Will be fixed in the next release later today.

closed 2025-02-11 by wimb0

wimb0, 2025-02-11:
Should be fixed in Release v2.3.2 
https://github.com/wimb0/home-assistant-saj-r5-modbus/releases/tag/v2.3.2