wimb0/home-assistant-saj-r5-modbus #111: Reload of the integration or restart of Home Assistant is nessesary every day.

Fritz4all, 2025-02-22:
Since the last update i need to reload the integration every morning to see solar production, i have 2 home assistnant installations, both wont show any Solar production untul a reload or complete reboot of home assistant.

Integration version : v2.3.2
Home Assistant : 2025.2.2 & 2025.2.5

koendel, 2025-02-23:
Yeah I also noticed this problem. Many error like these in the logs when the intergration crashes. 

`ConnectionResetError: [Errno 104] Connection reset by peer `
`BrokenPipeError: [Errno 32] Broken pipe `

Looks like the intergration can’t handle the interface going offline when the inverter starts and stops in the morning.

fsaris, 2025-02-23:
```
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.saj_modbus.hub
Bron: helpers/update_coordinator.py:380
integratie: SAJ R5 Inverter Modbus (documentatie, problemen)
Eerst voorgekomen: 14 februari 2025 om 17:34:36 (74659 gebeurtenissen)
Laatst gelogd: 10:25:39

Unexpected error fetching zonneplan data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/saj_modbus/hub.py", line 88, in _async_update_data
    realtime_data = await self.hass.async_add_executor_job(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.read_modbus_realtime_data
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/saj_modbus/hub.py", line 142, in read_modbus_realtime_data
    realtime_data = self._read_holding_registers(unit=1, address=0x100, count=60)
  File "/config/custom_components/saj_modbus/hub.py", line 68, in _read_holding_registers
    return self._client.read_holding_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 114, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=slave))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 203, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 104, in sync_execute
    self.pdu_send(request)
    ~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 160, in pdu_send
    self.low_level_send(self.trace_packet(True, packet), addr=addr)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/tcp.py", line 221, in send
    return self.socket.send(request)
           ~~~~~~~~~~~~~~~~^^^^^^^^^
BrokenPipeError: [Errno 32] Broken pipe
```

fsaris, 2025-02-23:
Looks a try catch is needed that closes/resets the connection when the pipe error happens.

Like https://github.com/joopert/nad_receiver/pull/59/files

Not sure if the connection is also reset on the ConnectionResetError

fsaris, 2025-02-23:
The connection reset error has the same origin 

```
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.saj_modbus.hub
Bron: helpers/update_coordinator.py:380
integratie: SAJ R5 Inverter Modbus (documentatie, problemen)
Eerst voorgekomen: 14 februari 2025 om 17:34:27 (2 gebeurtenissen)
Laatst gelogd: 22 februari 2025 om 18:05:32

Unexpected error fetching zonneplan data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/saj_modbus/hub.py", line 88, in _async_update_data
    realtime_data = await self.hass.async_add_executor_job(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.read_modbus_realtime_data
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/saj_modbus/hub.py", line 142, in read_modbus_realtime_data
    realtime_data = self._read_holding_registers(unit=1, address=0x100, count=60)
  File "/config/custom_components/saj_modbus/hub.py", line 68, in _read_holding_registers
    return self._client.read_holding_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 114, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=slave))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 203, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 108, in sync_execute
    return self.sync_get_response()
           ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 84, in sync_get_response
    if not (data := self.sync_client.recv(None)):
                    ~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/tcp.py", line 258, in recv
    if (recv_data := self.socket.recv(recv_size)) == b"":
                     ~~~~~~~~~~~~~~~~^^^^^^^^^^^
ConnectionResetError: [Errno 104] Connection reset by peer
```

wimb0, 2025-02-23:
I'm looking in to it.
It's weird that is is happening since the last version, must have something to do with either the pyModbus update or HA 2025.2 or something.
We already do some sort of error handling there.

```
    async def _async_update_data(self) -> dict:
        realtime_data = {}
        try:
            """Inverter info is only fetched once"""
            if not self.inverter_data:
                self.inverter_data = await self.hass.async_add_executor_job(
                    self.read_modbus_inverter_data
                )
            """Read realtime data"""
            realtime_data = await self.hass.async_add_executor_job(
                self.read_modbus_realtime_data
            )

            self.close()
        except ConnectionException:
            _LOGGER.error("Reading realtime data failed! Inverter is unreachable.")
            realtime_data["mpvmode"] = 0
            realtime_data["mpvstatus"] = DEVICE_STATUSSES[0]
            realtime_data["power"] = 0

        return {**self.inverter_data, **realtime_data}
```

I think the solution is to add BrokenPipeError and ConnectionResetError to the except.

Will try to fix later today or tomorrow.

(Good news btw, I've got myself an extra SAJ R5 inverter so I can test and debug the integration myself again in the near future. Still need to install the new PV array and inverter though.)

fsaris, 2025-02-23:
👍🏻

Shouldn't the `self.close()` be outside of the try catch? So always close/reset the connection when an error occurred

fsaris, 2025-02-25:
The integration was available this morning. So it works 😀👍🏻

wimb0, 2025-02-25:
Nice, thanks for testing!

wimb0, 2025-02-25:
Fixed in release v2.3.3.

closed 2025-02-25 by wimb0