wimb0/home-assistant-saj-r5-modbus #136: Question: limit power

JaySea77, 2025-05-10:
When I limit power to 0% the output is still 85W.
Is this normal behavior and can I turn it completely off to 0?

wimb0, 2025-05-11:
Interesting. I will look in to it.

If I set mine to 0%, the output drops to 0W according to raw data from the inverter for me, but it takes a few seconds.
Then the inverter goes to the "Waiting" state.
However, HA still reports 39W in my case.

I do see some bugs though, when the limit it at 0%, after a while it goes to NaN, and I can't adjust the slider anymore...

wimb0, 2025-05-11:
You can check the raw data from the inverter by using https://github.com/wimb0/saj_modbus

Runnning normal at 110%:
```
user@device:~/saj_modbus $ python3 read_r5_inverter_realtime_data.py  --host 192.168.x.x --port 502
{"mpvmode": 2, "mpvstatus": "Normal", "pv1volt": 189.6, "pv1curr": 6.59, "pv1power": 1252, "pv2volt": 0.0, "pv2curr": 655.35, "pv2power": 0, "pv3volt": 6553.5, "pv3curr": 655.35, "pv3power": 65535, "busvolt": 359.7, "invtempc": 27.5, "gfci": 1, "power": 1193, "qpower": 180, "pf": 0.989, "l1volt": 221.8, "l1curr": 5.22, "l1freq": 50.02, "l1dci": -2, "l1power": 1193, "l1pf": 0.989, "l2volt": 6553.5, "l2curr": 655.35, "l2freq": 655.35, "l2dci": -1, "l2power": 65535, "l2pf": -0.001, "l3volt": 6553.5, "l3curr": 655.35, "l3freq": 655.35, "l3dci": -1, "l3power": 65535, "l3pf": -0.001, "iso1": 32767, "iso2": 0, "iso3": 2654, "iso4": 0, "todayenergy": 1.38, "monthenergy": 127.05, "yearenergy": 594.95, "totalenergy": 1873.76, "todayhour": 2.9, "totalhour": 3419.5, "errorcount": 12, "datetime": "2025-05-11 09:49:37"}
```

When chaning the slider to 0%, it goes to a low value first:
```
user@device:~/saj_modbus $ python3 read_r5_inverter_realtime_data.py  --host 192.168.x.x --port 502
{"mpvmode": 2, "mpvstatus": "Normal", "pv1volt": 228.5, "pv1curr": 0.17, "pv1power": 43, "pv2volt": 0.0, "pv2curr": 655.35, "pv2power": 0, "pv3volt": 6553.5, "pv3curr": 655.35, "pv3power": 65535, "busvolt": 346.6, "invtempc": 29.1, "gfci": 0, "power": 39, "qpower": 0, "pf": 0.993, "l1volt": 217.3, "l1curr": 0.25, "l1freq": 49.97, "l1dci": -40, "l1power": 39, "l1pf": 0.993, "l2volt": 6553.5, "l2curr": 655.35, "l2freq": 655.35, "l2dci": -1, "l2power": 65535, "l2pf": -0.001, "l3volt": 6553.5, "l3curr": 655.35, "l3freq": 655.35, "l3dci": -1, "l3power": 65535, "l3pf": -0.001, "iso1": 32767, "iso2": 0, "iso3": 2664, "iso4": 0, "todayenergy": 1.35, "monthenergy": 127.02, "yearenergy": 594.92, "totalenergy": 1873.73, "todayhour": 2.8, "totalhour": 3419.4, "errorcount": 12, "datetime": "2025-05-11 09:49:48"}
```

After that the inverter goes to Waiting:
```
user@device:~/saj_modbus $ python3 read_r5_inverter_realtime_data.py  --host 192.168.x.x --port 502
{"mpvmode": 1, "mpvstatus": "Waiting", "pv1volt": 228.2, "pv1curr": 0.0, "pv1power": 41, "pv2volt": 0.0, "pv2curr": 655.35, "pv2power": 0, "pv3volt": 6553.5, "pv3curr": 655.35, "pv3power": 65535, "busvolt": 274.3, "invtempc": 29.1, "gfci": 1, "power": 0, "qpower": 0, "pf": 0.993, "l1volt": 216.5, "l1curr": 0.06, "l1freq": 49.99, "l1dci": 1, "l1power": 0, "l1pf": 0.993, "l2volt": 6553.5, "l2curr": 655.35, "l2freq": 655.35, "l2dci": -1, "l2power": 65535, "l2pf": -0.001, "l3volt": 6553.5, "l3curr": 655.35, "l3freq": 655.35, "l3dci": -1, "l3power": 65535, "l3pf": -0.001, "iso1": 32767, "iso2": 0, "iso3": 2664, "iso4": 0, "todayenergy": 1.35, "monthenergy": 127.02, "yearenergy": 594.92, "totalenergy": 1873.73, "todayhour": 2.8, "totalhour": 3419.4, "errorcount": 12, "datetime": "2025-05-11 09:50:00"}
```

JaySea77, 2025-05-11:
How can you turn the inverter back on when the slider isn't available?

Btw I used the integration that used the cloud service before. The values of the production per day, month and year were always available. Now when the inverter turns off, the values are unavailable. Is there a way to keep the values?

wimb0, 2025-05-11:
For me the slider became available after a few minutes. I could adjust it again.
It is definitely something that needs to be looked at. You are the first to report this.

For the values becoming unavailable: I have not found a reliable way to keep those values current from the integration, as the inverter is offline at night, so there is no data to be read. The only value that is "saved" is the total output, which is set to 0 when the inverter is offline. 
For the frontend, you could use template sensors to get the last state.

JaySea77, 2025-05-11:
I created a template sensor for this, but they don't survive a has reboot.

JaySea77, 2025-05-11:
Btw is there a way to change the polling interval without having to remove and add the device again?

wimb0, 2025-05-12:
I was able adjust the slider again after a few minutes, I just set it 110% again.

Currently, it is not possible to change the interval without having to remove and reinstall the add-on.

JaySea77, 2025-05-12:
Nice! I have it on every second at the moment. Btw I read somewhere that you do not have a Saj r5 anymore? How do you test then?

wimb0, 2025-05-12:
I have a SAJ R5 inverter again, so I can test on my own system.

The values for the interval are 10 to 600 seconds at the moment.
1 sec seems a bit too much, my inverter does not seem to like that many requests. If I set the interval to 1 sec, it sometimes take 2 to 3 seconds to get answer.
But if there is a need, I can change it to 1 sec as a minimum,

wimb0, 2025-05-12:
No problem.
I'll add 1 sec as a minimum. Should be available in V2.4.4 later today.

My P1 DSMR reader is also reporting every 1 second, but my automations to set the inverter limit only run every 10 seconds.

JaySea77, 2025-05-12:
I haven't automated the power limit yet, though I did make a switch with the scene to change the limit to either 110 when on or 0 when off. I can't rely on the power prices at the moment. The changes are not on the hour yet. Therefor I do it manually now.

wimb0, 2025-05-12:
A very simple and crude one:

If my usage or returned sensors are updated, do some maths to get the percentage of the max inverter output and set the limit to that number, wait 10 seconds for the next run of the automation.

Here is the YAML:

`p_max` is the output of your inverter. 2 in my case, because I a 2kW inverter.

```
description: ""
triggers:
  - entity_id:
      - sensor.dsmr_reading_electricity_currently_delivered
      - sensor.dsmr_reading_electricity_currently_returned
    from: null
    trigger: state
conditions: []
actions:
  - metadata: {}
    data:
      value: >-
        {% set p_max = 2 -%} {% set p_im =
        states('sensor.dsmr_reading_electricity_currently_delivered')|float(default=p_max) 
        -%} {% set p_ex =
        states('sensor.dsmr_reading_electricity_currently_returned')|float(default=p_max) 
        -%} {% set c =
        states('sensor.saj_active_power_of_inverter_total_output')|float(default=p_max
        * 1000) / 1000 -%} {% set s = p_max / 100 -%} {% set n = ((c - p_ex +
        p_im) / s) -%} {{ max( min(n, 110), 0)|round(0, 'ceil') }}
    target:
      entity_id: number.limit_power
    action: number.set_value
  - delay:
      hours: 0
      minutes: 0
      seconds: 10
      milliseconds: 0
mode: single
max_exceeded: silent

```

JaySea77, 2025-05-12:
This is to prevent sending back to grid?

wimb0, 2025-05-12:
Yes.
If for example my total usage is 200W according to the P1 meter , I limit the inverter to output only 200W.
That way I'm not sending back to the grid, but am generating enough to also not buy electicity from the grid.

JaySea77, 2025-05-23:
I noticed that when I set the limit_power in an automation, the entity doesn't seem to always show the new value. Though the power production is according the limit that was just set.

Let me know if you want me to do anything to give you more information.

wimb0, 2025-08-20:
> At least setting the limit to 0 makes it stay at 70W-100W at this moment. It doesn't se the throughput to 0 even with the 3.0.1

That is not something I can fix. By setting the slider to 0, the inverter does not turn off.
The inverter has no option to completely turn off.

> Edit: Setting a value seems to update my slider immediately to 0 or 110. I think it didn't always do that correctly before.

That is what should be fixed indeed. Adjusting the slider now should work OK.
I should also work better with automations.

wimb0, 2025-08-26:
> I created the power switch myself, but that only changed the power limit to 0 or 110. This power switch is a lot better.
> 
> Btw I edited my reply after you replied.

Good to hear. It took a long time, and just stumbled upon this setting in the newer documentation.

wimb0, 2025-08-26:
I noticed that too. I think that is the inverter calibrating itself and doing checks. Turning on takes a while. Generating power even longer. That is just how the thing works I guess.

I'll close this issue, but feel free to comment more if needed. Or open a new issue if needed.

closed 2025-08-26 by wimb0

JaySea77, 2025-08-26:
I guess this should work? Since the saj is offline now, the saj_powe_on_off switch is unavailable.

alias: Zonnepanelen uit op prijs
description: ""
triggers:
  - trigger: numeric_state
    entity_id:
      - sensor.dynamische_prijzen_current_electricity_market_price
    below: 0
conditions:
  - condition: state
    entity_id: switch.saj_power_on_off
    state: "on"
actions:
  - type: turn_off
    device_id: xxxxx
    entity_id: xxxx
    domain: switch
  - action: notify.mobile_app_iphone_van_jaysea77
    data:
      message: Zonnepanelen uitgezet ivm negatieve prijs
      title: Zonnepanelen
mode: single