wimb0/home-assistant-saj-r5-modbus #137: Sensor become unknown intermittently on HA 2025.5.1

ErikT80, 2025-05-11:
Using the SAJ integration with a lot of pleasure but noticed that after HA 2025.1 update the sensor become unknown intermittently. It seems that every other fetch the data is updated but the fetch after results in unknown sensors. 

Already tried with various polling intervals (5s, 20s, 60s) but with the same result.

Only thing I get from the debug logs are:

ERROR (MainThread) [custom_components.saj_modbus.hub] Reading realtime data failed! Inverter is unreachable.

I checked my EW11 adapter and it has a wifi RSSI of 100 and is available all the time.

wimb0, 2025-05-11:
I cannot reproduce this issue on my own install with HA 2025.1.
My polling is at 30s. Tested also at 60s.

Do you have any other modbus integrations active?
I the past there were some issues when other integrations that use modbus.
Those were causing issues for other integrations after a HA update.

ErikT80, 2025-05-11:
Hmm ok, after a restart of HA it worked again for 10 hour after which it started to show the same behaviour again. I now tried to completely remove the integration, reinstall and add the integration again. 

I have no other integration that use modbus, but have to say I deleted and added the saj a couple of times the last days to change the polling interval (or is there a smarter way?).

Will keep you posted if reinstalling worked..

ErikT80, 2025-05-12:
Update: issue is still there after a reinstall. I have my polling interval at 10 sec. Could that be an issue? It feels to me that after a HA restart it works for 10-12 hours.

wimb0, 2025-05-12:
Is the EW11 the only device connected to your inverter?
With a short interval, it could be that you are getting interference while reading data from the other device.

ErikT80, 2025-05-12:
Also have my zonneplan dongle connected.. will disconnect it and see if it will improve. Thanks for the hint :-)

ErikT80, 2025-05-12:
No luck.. still the same issue 9-10 hours after HA restart. After restart everything is fine again. Will try again with polling interval at 30 seconds.

wimb0, 2025-05-12:
I've release version V2.4.3, which adds a configure button to change the interval without reinstalling the addon.

ErikT80, 2025-05-13:
Thanks for the release! Really useful! Unfortunately after my 30 polling interval test it still the sensors intermittently  became unknown after roughly 9-10 hours after my HA restart.

@wimb0 any idea how to debug this issue? Since it works after a HA restart I suspect it has nothing to do with the EW11 or the inverter..

wimb0, 2025-05-14:
Maybe you could enable debugging for modbus in HA:
I think that only works in configuration.yaml, and needs a HA restart:

```
  logs:
    homeassistant.components.modbus: debug
```

ErikT80, 2025-05-14:
Thanks, just to be sure. I am running Home assistant OS with pymodbus version 3.8.3. Which version do you have @wimb0?

I also see other (kinda) related issues on modbus in home assistant core:

[https://github.com/home-assistant/core/issues/140580](Modbus integration gets stuck every few days)

wimb0, 2025-05-16:
I also use Home Assistant OS, the current pymodbus version is 3.8.3

I'm still not able to reproduce the issue :(

did you get any modbus debug logs?

ErikT80, 2025-05-18:
Well didn't get any useful logs.. since I enabled modbus logging plus debug logging on this integration I haven't seen the issue anymore... Really strange because I didn't change anything on my setup. But glad it is working again.

wimb0, 2025-05-18:
Maybe the issue was in HA itself, and fixed in 2025.5.1 or 2025.5.2?

I'll close this issue, since it is working for you again.

closed 2025-05-18 by wimb0