wimb0/home-assistant-saj-r5-modbus #145: Unexpected name change saj_limit_power to limit_power

gitproos, 2025-05-13:
After updating to v2.4.4 there was an unexpected name change:

Previously the setting for power limit was called:
number.saj_limit_power

After the update it changed to:
number.limit_power

This minor change breaks existing existing scripts and automations; as I automatically tune the limit to follow demand.

Although it is easily fixed I am not sure if this change was intentional, as it was not mentioned in the release notes and other variables names still include the 'saj' part. If it was intentional I will happily update the scripts at my end :)

wimb0, 2025-05-13:
That is interesting. For me it has been `number.limit_power` since at least v2.3.0.
In the last few versions nothing was changed about this.

Do you know in which version it was correctly named `number.saj_limit_power` ?

gitproos, 2025-05-13:
I am following default updates, so I suspect v2.4.3. It changed the name and disabled temperature sensor after installing the update last night. Noticed this after the sun came up.

As limit_power is expected I will proceed to change the scripts 👍

gitproos, 2025-05-13:
Thank you!

As a side note: it is very convenient now to change the polling frequency via configuration. That is a nice feature. Although I noticed it showed 60 seconds (default?) instead of 8 or 10 seconds I had previously. Changed that and I've update my scripts and everything is working fine now. I'll close the issue!

wimb0, 2025-05-13:
Glad to hear. Correct. Unfortunately that is also because of the config version change. Most sensors and the polling frequency are at default again. 
Sorry for the inconvenience, I hoped those settings would be migrated over, but that doenst happen.