wimb0/home-assistant-saj-r5-modbus #176: SAJ R5 Inverter Modbus Stops working after upgrading to Home Assistant Core 2025.9.0b1 & 2025.9.0b2

Fritz4all, 2025-09-01:
Working Versions : 

      Installation method        : Home Assistant OS
      Core                                : 2025.8.3
      Supervisor                       : 2025.08.3
      Operating System           : 16.2.rc1
      Frontend                         : 202508.11.1
      SAJ R5 Inverter Modbus : v3.1.**0**
      
      Installation method        : Home Assistant OS
      Core                                : 2025.8.3
      Supervisor                       : 2025.08.3
      Operating System           : 16.2.rc1
      Frontend                         : 202508.11.1
      SAJ R5 Inverter Modbus : v3.1.**1**

Not working Version  :

      Installation method        : Home Assistant OS
      Core                                : 2025.9.0b1 & 2025.9.0b2
      Supervisor                       : 2025.08.3
      Operating System           : 16.2.rc1
      Frontend                         : 20250828.0
      SAJ R5 Inverter Modbus : v3.1.1

Error : 

    [31m2025-09-01 13:14:35.885 ERROR (MainThread) [custom_components.saj_modbus.hub] Unexpected error fetching SAJ data
    Traceback (most recent call last):
      File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 331, in __wrap_async_setup
        await self._async_setup()
      File "/config/custom_components/saj_modbus/hub.py", line 51, in _async_setup
        self.inverter_data = await self.hass.async_add_executor_job(
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
            self.read_modbus_inverter_data
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        )
        ^
      File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
        result = self.fn(*self.args, **self.kwargs)
      File "/config/custom_components/saj_modbus/hub.py", line 122, in read_modbus_inverter_data
        inverter_data = self._read_holding_registers(unit=1, address=0x8F00, count=29)
      File "/config/custom_components/saj_modbus/hub.py", line 90, in _read_holding_registers
        return self._client.read_holding_registers(
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
            address=address, count=count, slave=unit
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        )
        ^
    TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'[0m

wimb0, 2025-09-01:
I noticed this too in my testing.
In HA 2025.09, pymodbus is updated to 3.11.1.
Unfortunately this contains a backwards incompatible change:
See: https://pymodbus.readthedocs.io/en/latest/source/changelog.html#version-3-10-0

wimb0, 2025-09-01:
I have released pre-release version v3.1.2, which containts the fix.